moveit/moveit_setup_assistant #102: Released configuration pkgs cannot create db in default location (demo.launch with db:=true)

gavanderhoorn, 2015-01-12:
Launching `demo.launch db:=true` (from `pr2_moveit_config` fi) results in:
```
Traceback (most recent call last):
  File "/opt/ros/hydro/lib/warehouse_ros/mongo_wrapper_ros.py", line 93, in <module>
    os.makedirs(dbpath)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 13] Permission denied: '/opt/ros/hydro/share/pr2_moveit_config/default_warehouse_mongo_db'
```

Looking in `default_warehouse_db.launch` it seems the `moveit_warehouse_database_path` is set to `$(find pr2_moveit_config)/default_warehouse_mongo_db`, which for released packages resolves to `/opt/ros/$DISTRO/share/pr2_moveit_config/default_warehouse_mongo_db`. As that is non-writable for a normal user, that results in a *Permission denied* error.

I'm not sure whether this is by-design (and I'm just misunderstanding things): it could be that the intended use of `default_warehouse_db.launch` is to be just that: a default. If users want to make use of the db, they should start `warehouse.launch` with a suitable value for `moveit_warehouse_database_path`. However, `demo.launch` exposes the `db` parameter currently, which leads to the described behaviour, which seems like something that needs to be fixed.

gavanderhoorn, 2015-01-12:
Provided I understand things correctly: would it be an option to make `moveit_warehouse_database_path` configurable from outside `default_warehouse_db.launch`? With the default being the current `$(find [GENERATED_PACKAGE_NAME])/default_warehouse_mongo_db`?

That way `demo.launch` could provide access to the parameter.

davetcoleman, 2015-01-13:
Yes, this is a bug that devs haven't run into because we always build from source.

> make moveit_warehouse_database_path configurable from outside default_warehouse_db.launch?

That is a good solution, feel free to send a PR!

commit 35ef150d (2015-01-21): Update warehouse launch file to accept non-standard db location. Fix #102.

Also update generated demo.launch accordingly.

The default directory could be located on a non-writable file system, leading
to crashes of the mongodb wrapper script. This change allows the user to specify
an alternative location using the 'db_path' argument.
closed 2015-01-21 by isucan