moveit/moveit_setup_assistant #106: New release?

gavanderhoorn, 2015-05-23:
Just wondering if it would be possible to do a release of `moveit_setup_assistant` into Indigo/Jade to get everything [past 0.6.0](https://github.com/ros-planning/moveit_setup_assistant/compare/0.6.0...indigo-devel) released?

#102 keeps biting me on machines that don't have the assistant from-source.

gavanderhoorn, 2015-09-04:
Friendly ping?

(reminded by ROS Answers: [MoveIt Setup Assistant ROS Jade](http://answers.ros.org/question/217049/moveit-setup-assistant-ros-jade/))

gavanderhoorn, 2016-03-04:
Fixed with 8da76a5f.

alextoind, 2016-03-07:
@gavanderhoorn actually the fix is only for Indigo. By compiling the source it works pretty fine in Jade, but it would be cleaner if `ros-jade-moveit-setup-assistant` is released.

gavanderhoorn, 2016-03-07:
You're right @alextoind: I've changed the title of the issue and reopened it.

130s, 2016-06-24:
Working on this, finally.

gavanderhoorn, 2016-06-24:
Much appreciated, thanks :tada:

130s, 2016-06-24:
I'll wait at least for https://github.com/ros-planning/moveit_setup_assistant/issues/110 (that has a PR to) to be resolved.

davetcoleman, 2016-06-24:
I don't think you need to wait for #110 - it's not a critical bug, just a suggested new feature. The behavior has never been an issue for the ~5 years since I wrote the Setup Assistant until recently. Of course it would be a nice improvement when its ready.

130s, 2016-06-24:
Oh, kay. Making a Jade release now.

130s, 2016-06-24:
Done https://github.com/ros/rosdistro/pull/11975

closed 2016-06-24 by 130s