moveit/moveit_setup_assistant #110: Moveit setup assistant fills in invalid author and maintainer emails

tfoote, 2016-01-27:
Auditing packages there are a lot of them with Author and Maintainer emails of assistant@moveit.ros.org

The tool should not fill in this as the Author and Maintainer in the generated package.xml it's important that these values get filled in as the actual author and maintainer so they can get notifications. It should probably prompt you to help get the right values or at least put todos in so it's obvious to users that it's not filled, like catkin_create_pkg

davetcoleman, 2016-02-04:
Here's a stop-gap, based on how catkin_create_pkgs does it:
https://github.com/ros-planning/moveit_setup_assistant/pull/111

mikeferguson, 2016-08-26:
replaced by https://github.com/ros-planning/moveit/issues/16

closed 2016-08-26 by mikeferguson