moveit/moveit_setup_assistant #119: Segfault while launching moveit_setup_assistant on jade-devel on ROS Kinetic

dbolkensteyn, 2016-07-26:
Launching `moveit_setup_assistant` using the following command [as documented](http://docs.ros.org/jade/api/moveit_setup_assistant/html/doc/tutorial.html) on ROS Kinetic yields a segmentation fault.
```
roslaunch moveit_setup_assistant setup_assistant.launch
```

output:
```
SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.2

NODES
  /
    moveit_setup_assistant (moveit_setup_assistant/moveit_setup_assistant)

auto-starting new master
process[master]: started with pid [17777]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ed290684-5319-11e6-b625-0242ac110002
process[rosout-1]: started with pid [17790]
started core service [/rosout]
process[moveit_setup_assistant-2]: started with pid [17802]
================================================================================REQUIRED process [moveit_setup_assistant-2] has died!
process has died [pid 17802, exit code -11, cmd /root/ws_moveit/devel/lib/moveit_setup_assistant/moveit_setup_assistant __name:=moveit_setup_assistant __log:=/root/.ros/log/ed290684-5319-11e6-b625-0242ac110002/moveit_setup_assistant-2.log].
log file: /root/.ros/log/ed290684-5319-11e6-b625-0242ac110002/moveit_setup_assistant-2*.log
Initiating shutdown!
================================================================================
[moveit_setup_assistant-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

There is no such file as `/root/.ros/log/ed290684-5319-11e6-b625-0242ac110002/moveit*`.

I've built MoveIt myself from the [experimental unified repository](https://github.com/davetcoleman/moveit/).

I've tried to add a breakpoint on the entry of the `main` function in `setup_assistant_main.cpp` using CLion [I'm not familiar with gdb] - and it does not even get hit.
```
long double boost::math::detail::erf_imp<long double, boost::math::policies::policy<boost::math::policies::promote_float<false>, boost::math::policies::promote_double<false>, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy> >(long double, bool, boost::math::policies::policy<boost::math::policies::promote_float<false>, boost::math::policies::promote_double<false>, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy> const&, mpl_::int_<64> const&)
```
[image]

I've googled a bit, and found that such Boost errors might be caused by different versions or flags of the compiler. I've first tried to build MoveIt with clang, which failed, and then tried to use the already-built from sources [moveit-kinetic-source Docker image](https://github.com/ros-planning/moveit_docker). Even on that image, I'm getting the same segmentation fault upon the start of the `moveit_setup_assistant ` node.

dbolkensteyn, 2016-07-26:
FYI : I'm going to try to run a `git bisect` on the unified repo to see if I can go back to a commit where this doesn't happen - but I'll only test against ROS Kinetic

dbolkensteyn, 2016-07-26:
Looks like there was an elephant in the room: A very recent upgrade to C++11.

I was able to fix this issue by reverting it [see https://github.com/davetcoleman/moveit/pull/7] - but I am not familiar enough with C++ and Boost's black magic to understand why.

This seems to relate to https://groups.google.com/forum/#!topic/moveit-users/F9vl9OadFf8

dbolkensteyn, 2016-07-26:
According to https://github.com/PointCloudLibrary/pcl/issues/619 and https://github.com/felixendres/rgbdslam_v2/issues/8, using C++11 is all in or all out: You can't safely mix libraries compiled with different versions of the standard.

v4hn, 2016-07-26:
We are currently working on moving MoveIt! to c++11 in *all* subprojects.
So you will have to wait for a bit or help us out there to get it to work in kinetic.
See https://github.com/ros-planning/moveit_core/issues/315 for details

dbolkensteyn, 2016-07-26:
This duplicates https://github.com/ros-planning/moveit_core/issues/313

closed 2016-07-26 by dbolkensteyn

davetcoleman, 2016-07-26:
@dbolkensteyn can you open a PR that switches the setup assistant to C++11? To get Travis to pass you'll also need to switch the .travis.yml to moveit_ci https://github.com/ros-planning/moveit_ci